leoheck/kiri #111: Cannot view board layout

ahalekelly, 2024-03-13:
I just installed Kiri and cannot view the board layouts of any projects. I was able to reproduce this with a public project, [megadesk](https://github.com/gcormier/megadesk/), here's the final section of the `kiri -V` output:

```
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

# 22/231 8f813aa | 2022-11-30 21:15:04 | Greg Cormier | change folder case

    # Rev 8f813aa files
         project_path: /Users/akelly/Downloads/megadesk/pcb/.kiri/8f813aa
              project: megadesk.kicad_pro
            schematic: megadesk.kicad_sch
               layout: megadesk.kicad_pcb

    # Rev 8f813aa project info
        Kicad Version: 6
          Sch Version: 20211123
          PCB Version: 20211014

    # Plotting schematics, 8f813aa
    DIR: /Users/akelly/Downloads/megadesk/pcb
    kicad-cli sch export svg --black-and-white --no-background-color --output "/Users/akelly/Downloads/megadesk/pcb/.kiri/8f813aa/_KIRI_/sch" \
    "/Users/akelly/Downloads/megadesk/pcb/.kiri/8f813aa/megadesk.kicad_sch"
    status: 0

    # Plotting layouts 4a2ee6c, 8f813aa
    DIR: /Users/akelly/Downloads/megadesk/pcb
    kidiff -k -n --webserver-disable -a 4a2ee6c -b 8f813aa -o "/Users/akelly/Downloads/megadesk/pcb/.kiri" -f "megadesk.kicad_pcb"
    | ./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
    | Traceback (most recent call last):
    |   File "/Users/akelly/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 264, in <module>
    |     processBoard(board_path, plot_dir, args.quiet, args.verbose, args.frame, args.numbers)
    |   File "/Users/akelly/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 119, in processBoard
    |     popt.SetDrillMarksType(pn.PCB_PLOT_PARAMS.NO_DRILL_SHAPE)
    | AttributeError: type object 'PCB_PLOT_PARAMS' has no attribute 'NO_DRILL_SHAPE'
    |
    | ./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
    | Traceback (most recent call last):
    |   File "/Users/akelly/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 264, in <module>
    |     processBoard(board_path, plot_dir, args.quiet, args.verbose, args.frame, args.numbers)
    |   File "/Users/akelly/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 119, in processBoard
    |     popt.SetDrillMarksType(pn.PCB_PLOT_PARAMS.NO_DRILL_SHAPE)
    | AttributeError: type object 'PCB_PLOT_PARAMS' has no attribute 'NO_DRILL_SHAPE'
    |
    status: 1

    # Rev pcb files
         project_path: /Users/akelly/Downloads/megadesk/pcb
              project: megadesk.kicad_pro
            schematic: megadesk.kicad_sch
               layout: megadesk.kicad_pcb

Assembling the HTML
- Adding list of commits
- Adding list of pages
- Adding list of layers

Latest rev 4a2ee6c
- Project's title (folder name): megadesk
- Project's name (file name): megadesk
- Sch title: [missing]
- PCB title: [missing]
- Sch revision: [missing]
- PCB revision: [missing]
- Sch date: [missing]
- PCB date: [missing]

Kiri main page
- Initial project path:
  Pro 1: .kiri/4a2ee6c/megadesk.kicad_pro
  Pro 2: .kiri/f710b6e/megadesk.kicad_pro

Starting webserver at http://127.0.0.1:8080/web/index.html
(Hit Ctrl+C to exit)
```

Screenshot:
[image]

System specs:
Kicad 8.0 from homebrew
MacOS 14.2.1
Apple Silicon M1

leoheck, 2024-03-13:
I ran kiri on megadesk quickly here, and the layout did not work for me on Linux either.
Maybe I have to improve some paths of Python, I will check that. 

However, I could see the layout running this it with the -k flag like this. This uses `kicad-cli` instead of `kicad-diff` to plot layouts, but it takes more time. So I suggest using -t to limit the N last commits like this:
```
kiri -r -D -t 4 -k
```

Also, it this date is a problem, you may not have GNU Date. Check if you have this `gdate` program.
If you have it, use this workaround while studying how to fix it.
```
alias date=gdate
```

In summary you can try this to check if Kiri works complete again for you on MacOS:
```
alias date=gdate
kiri -r -D -k -t 4
```

Please, let me know if this fixes the issue for you while I track and solve this.

ahalekelly, 2024-03-13:
Thanks for the quick response! `kiri -r -D -t 4 -k` works, it seems to plot the sheet border too though, I'm not sure if that's intended behavior:
[image]

The `date` command doesn't seem to be a problem anywhere except for `kiri -v`. gdate version 9.4 is installed on my system, but running `alias date=gdate` doesn't fix `kiri -v`

leoheck, 2024-03-13:
it is the intended behavior. there might a flag to remove that. don't
remember if it will do something with the -k flag. Check it with -h, please.

Also, if you zoom in the view should remain as it is...

lowie2727, 2024-04-02:
I cannot switch to the `Layout View` tab, the button in the web page does nothing, it just stays in the schematic. Did the update and it's still broken.

leoheck, 2024-04-04:
Hi @lowie2727  it looks like your repo does not have layout, or there is a problem somewhere.

You have 2 options to check this.

1. First, try to use the `-r` flag to remove previous runs.
```
kiri -r -D -t 4
```

2. Use the `-k` to use `kicad-cli` instead of `kidiff` to plot the layout. This takes a bit longer than using `kidiff`
```
kiri -r -D -k -t 4
```

If none of these work, check the command line output when using `-D` flag to see if there is something bad happening to the command that plots the layouts.

lowie2727, 2024-04-04:
Hi @leoheck 

This is the output i get when running `kiri -r -D -t 4`
---

```zsh
Using Kicad project ./microSD.kicad_pro

          KIRI_HOME_PATH = /home/lowie/.local/share/kiri
           KICAD_VERSION = 8.0.1-1.fc39
            CURRENT_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd
 KICAD_PRO_ABSOLUTE_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/microSD.kicad_pro
      REPO_ABSOLUTE_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd
  KICAD_PROJ_NESTED_PATH = .
KICAD_PROJ_ABSOLUTE_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
         OUTPUT_DIR_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri
            WEB_DIR_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/web/

    # Rev local files
         project_path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
              project: microSD.kicad_pro
            schematic: microSD.kicad_sch
               layout: microSD.kicad_pcb
```
```zsh
Output folder: .kiri
Removing leftovers from the previous run
Get commits command
   git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | %ad | %an | %s' main -- microSD.kicad_sch microSD.kicad_sch microSD.kicad_pcb

Checking local changes
[ ] ./microSD.kicad_pcb
[ ] ./microSD.kicad_sch
Files have local changes? No

Commits list
     1	33bcf51    2024-04-03 16:11:32    lowie2727     changed copper finish + solder mask and silkscreen color update
     2	9b04313    2024-04-03 15:29:13    lowie2727     updated polygon
     3	b846bdb    2024-04-03 15:19:18    lowie2727     removed unused copper layers
     4	776ca78    2024-04-02 00:41:06    KatoWarson    silkscreen + BOM update
     5	81a828f    2024-04-02 00:27:30    lowie2727     silkscreen update
     6	da18d67    2024-03-14 01:15:11    lowie2727     minor fixes schematic and PCB
     7	537d29a    2024-03-07 13:42:15    lowie2727     added info to generate BOM file
     8	ac22bc5    2024-03-07 02:43:18    lowie2727     fully migrated to KiCad
     9	f8df56a    2024-03-06 13:57:44    lowie2727     added drawing sheet
    10	7ab7bdb    2024-03-05 20:37:33    lowie2727     Eagle import
    11	227a5f6    2024-03-05 20:22:20    lowie2727     blank KiCad project

Artifacts Summary
11 commits (55 comparisons)

Generating artifacts (may take some time)
```

```zsh
# 1/55 33bcf51 | 2024-04-03 16:11:32 | lowie2727 | changed copper finish + solder mask and silkscreen color update
# 2/55 9b04313 | 2024-04-03 15:29:13 | lowie2727 | updated polygon

    Retrieving commits:
    - Source path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
    - Output path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51

    Retrieving commits:
    - Source path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
    - Output path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313

    # Rev 33bcf51 files
         project_path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51
              project: microSD.kicad_pro
            schematic: microSD.kicad_sch
[... 158 lines cut ...]

    Sch: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/microSD.kicad_sch
    Pages: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/sch_sheets

    PCB: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/microSD.kicad_pcb
    Layers: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/pcb_layers

    # Plotting schematics, 33bcf51
    DIR: /home/lowie/git/solar cooker/kicad-adafruit-microsd
    kicad-cli sch export svg --black-and-white --no-background-color --output "/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51/_KIRI_/sch" "/home/lowie/git/solar \
    cooker/kicad-adafruit-microsd/.kiri/33bcf51/microSD.kicad_sch"
    | Plotted to '/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51/_KIRI_/sch/microSD.svg'.
    | Done
    status: 0

    # Plotting schematics, 9b04313
    DIR: /home/lowie/git/solar cooker/kicad-adafruit-microsd
    kicad-cli sch export svg --black-and-white --no-background-color --output "/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/sch" "/home/lowie/git/solar \
    cooker/kicad-adafruit-microsd/.kiri/9b04313/microSD.kicad_sch"
    | Plotted to '/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/sch/microSD.svg'.
    | Done
    status: 0
```

It looks like when using the kicad-cli it doens't include the PCB files at all. It is only plotting schematics. Without the -k flag it looks like it is plotting the PCB without any errors but i just can't see it in the browser. It tried using a diffrent browser but no succes. Hope the outputs help a bit, let me know.

leoheck, 2024-04-04:
This log shows nothing wrong.
What does the `kiri -v` show?
Did you upgrade kiri to the latest commit? Can you do that?
Could you `ls` the project folder and also `pwd` inside of it?

leoheck, 2024-04-04:
Meanwhile, can you test Kiri using this repo here? 
Please, to help us don't use spaces on the path while testing this.
Spaces on the filepath should work, but let's avoid this problem during this test.
https://github.com/leoheck/assoc-board

lowie2727, 2024-04-04:
kiri -v output

```zsh
    kiri aead11b 2024/04/04 19h46
    kicad 8.0.1-1.fc39
    plotgitsch v0.9.0-8-ge5f94e4
    kidiff 9c4b17a 2024/04/02 20h25
```

ls output

```zsh
drwxr-xr-x@    - lowie lowie  4 Apr 19:56 .git
drwxr-xr-x@    - lowie lowie  4 Apr 16:53 .kiri
drwxr-xr-x@    - lowie lowie 14 Mar 17:22 build
drwxr-xr-x@    - lowie lowie 12 Mar 17:51 external
drwxr-xr-x@    - lowie lowie 14 Mar 17:33 microSD-backups
.rw-r--r--@  425 lowie lowie 12 Mar 17:51 .gitignore
.rw-r--r--@ 2.6k lowie lowie 12 Mar 17:51 drawing_sheet.kicad_wks
.rw-------@ 3.4M lowie lowie 14 Mar 17:00 fp-info-cache
.rw-r--r--@  142 lowie lowie 12 Mar 17:51 fp-lib-table
.rw-r--r--@   11 lowie lowie 12 Mar 17:51 microSD.kicad_dru
.rw-r--r--@ 211k lowie lowie  4 Apr 16:42 microSD.kicad_pcb
.rw-r--r--@ 1.3k lowie lowie 12 Mar 18:43 microSD.kicad_prl
.rw-r--r--@  15k lowie lowie  4 Apr 16:42 microSD.kicad_pro
.rw-r--r--@  79k lowie lowie  4 Apr 16:42 microSD.kicad_sch
.rw-r--r--@  177 lowie lowie 12 Mar 17:51 README.md
.rw-r--r--@  164 lowie lowie 12 Mar 17:51 sym-lib-table
```

pwd

```
/home/lowie/git/solar-cooker/kicad-adafruit-microsd
```

i removed the space from the folder name
I'll checkout the assoc-board.

lowie2727, 2024-04-04:
The assoc-board is working fine. Kiri detects all layers of this board which is not the case with [my project](https://github.com/solar-cooker-UHasselt/kicad-adafruit-microsd). The project is imported from Eagle so mabye the layers differ.

output kiri assoc-board:

```zsh
# 1/66 a7bb19c | 2024-04-02 15:07:25 | Leandro Heck | Update to Kicad 8
# 2/66 7aebd2e | 2023-03-22 16:09:20 | Leandro Heck | Update to Kicad 7

    Retrieving commits:
    - Source path: /home/lowie/git/assoc-board/.
    - Output path: /home/lowie/git/assoc-board/.kiri/a7bb19c

    Retrieving commits:
    - Source path: /home/lowie/git/assoc-board/.
    - Output path: /home/lowie/git/assoc-board/.kiri/7aebd2e

    # Rev a7bb19c files
         project_path: /home/lowie/git/assoc-board/.kiri/a7bb19c
              project: board.kicad_pro
            schematic: board.kicad_sch
            cache.lib: board-cache.lib
               layout: board.kicad_pcb

    # Rev a7bb19c project info
        Kicad Version: Unknown
          Sch Version: kicad_sch
          PCB Version: kicad_pcb

    Sch: /home/lowie/git/assoc-board/.kiri/a7bb19c/board.kicad_sch
    Pages: /home/lowie/git/assoc-board/.kiri/a7bb19c/_KIRI_/sch_sheets

    PCB: /home/lowie/git/assoc-board/.kiri/a7bb19c/board.kicad_pcb
    Layers: /home/lowie/git/assoc-board/.kiri/a7bb19c/_KIRI_/pcb_layers

    # Rev 7aebd2e files
         project_path: /home/lowie/git/assoc-board/.kiri/7aebd2e
              project: board.kicad_pro
            schematic: board.kicad_sch
            cache.lib: board-cache.lib
               layout: board.kicad_pcb

    # Rev 7aebd2e project info
        Kicad Version: 7
          Sch Version: 20230121
          PCB Version: 20221018

    Sch: /home/lowie/git/assoc-board/.kiri/7aebd2e/board.kicad_sch
    Pages: /home/lowie/git/assoc-board/.kiri/7aebd2e/_KIRI_/sch_sheets

    PCB: /home/lowie/git/assoc-board/.kiri/7aebd2e/board.kicad_pcb
    Layers: /home/lowie/git/assoc-board/.kiri/7aebd2e/_KIRI_/pcb_layers

    # Plotting schematics, a7bb19c
    DIR: /home/lowie/git/assoc-board
    kicad-cli sch export svg --black-and-white --no-background-color --output "/home/lowie/git/assoc-board/.kiri/a7bb19c/_KIRI_/sch" \
    "/home/lowie/git/assoc-board/.kiri/a7bb19c/board.kicad_sch"
    | Plotted to '/home/lowie/git/assoc-board/.kiri/a7bb19c/_KIRI_/sch/board.svg'.
    | Done
    status: 0

    # Plotting schematics, 7aebd2e
    DIR: /home/lowie/git/assoc-board
    kicad-cli sch export svg --black-and-white --no-background-color --output "/home/lowie/git/assoc-board/.kiri/7aebd2e/_KIRI_/sch" \
    "/home/lowie/git/assoc-board/.kiri/7aebd2e/board.kicad_sch"
    | Plotted to '/home/lowie/git/assoc-board/.kiri/7aebd2e/_KIRI_/sch/board.svg'.
    | Done
    status: 0

    # Plotting layout 7aebd2e, board.kicad_pcb - F.Cu
    DIR: /home/lowie/git/assoc-board
[... 149 lines cut ...]

    # Plotting layout 7aebd2e, board.kicad_pcb - F.CrtYd
    DIR: /home/lowie/git/assoc-board
    kicad-cli pcb export svg --page-size-mode 0 --black-and-white "/home/lowie/git/assoc-board/.kiri/7aebd2e/board.kicad_pcb" --output \
    "/home/lowie/git/assoc-board/.kiri/7aebd2e/_KIRI_/pcb/layer-47.svg" --layers "Edge.Cuts,F.CrtYd"
    | Loading board
    | Successfully created svg file
    status: 0

    # Plotting layout 7aebd2e, board.kicad_pcb - B.Fab
    DIR: /home/lowie/git/assoc-board
    kicad-cli pcb export svg --page-size-mode 0 --black-and-white "/home/lowie/git/assoc-board/.kiri/7aebd2e/board.kicad_pcb" --output \
    "/home/lowie/git/assoc-board/.kiri/7aebd2e/_KIRI_/pcb/layer-48.svg" --layers "Edge.Cuts,B.Fab"
    | Loading board
    | Successfully created svg file
    status: 0

    # Plotting layout 7aebd2e, board.kicad_pcb - F.Fab
    DIR: /home/lowie/git/assoc-board
    kicad-cli pcb export svg --page-size-mode 0 --black-and-white "/home/lowie/git/assoc-board/.kiri/7aebd2e/board.kicad_pcb" --output \
    "/home/lowie/git/assoc-board/.kiri/7aebd2e/_KIRI_/pcb/layer-49.svg" --layers "Edge.Cuts,F.Fab"
    | Loading board
    | Successfully created svg file
    status: 0
```

leoheck, 2024-04-04:
Ok, I may have found something.. 

Can you open the PCB and the main schematic of your repo with the text editor and give me the version fo your files?

```
(kicad_sch
	(version 20231120)
```

```
(kicad_pcb
	(version 20240108)
```

leoheck, 2024-04-04:
Ah, you have shared your project. Good, let me check it.

leoheck, 2024-04-04:
Ok, I could find the issue. I just don't know how to fix this yet.
But here is a preview on this working after running it with kriri.
It worked but I cannot commit it yet since the solution is something that is not quite good.
[image]

leoheck, 2024-04-04:
ok, I could push some code. can you test it? I think it is going to work for you, hope I did not mess up for others.
Please, let me know if this is working. Cool?

lowie2727, 2024-04-04:
Hi
Thank you very much, it works now! No complaints at the moment. I appreciate the work you do. You can close the issue now ;).

closed 2024-04-04 by leoheck