leoheck/kiri #112: Unable to view schematic or layout diff

rishabhrkaushik, 2024-03-29:
I've a similar issue as #111. I'm unable to view schematic or board diff. I'm using Win 11 and running Kiri through WSL2 Ubuntu 22.04.3. I tried the solution recommended in #111 of using -r -D -t 2 -k flag without any success. 

My hypothesis is it has something to do with version of the Kicad as it shows 6.0.2 but my project version is 7 and lately updated to 8 (i was having same issue before upgrading as well)

Below is a part of the log statements which some information about the troubles. 

`
KIRI_HOME_PATH = /home/rishabh/.local/share/kiri
KICAD_VERSION = 6.0.2+dfsg-1
CURRENT_PATH = /home/rishabh
KICAD_PRO_ABSOLUTE_PATH = /mnt/d/Work/Projects/CustomMacropad/PCB/TiledFlexiKeys/TiledFlexiKeys.kicad_pro
REPO_ABSOLUTE_PATH = /mnt/d/Work/Projects/CustomMacropad/PCB
KICAD_PROJ_NESTED_PATH = TiledFlexiKeys
KICAD_PROJ_ABSOLUTE_PATH = /mnt/d/Work/Projects/CustomMacropad/PCB/TiledFlexiKeys
OUTPUT_DIR_PATH = /mnt/d/Work/Projects/CustomMacropad/PCB/TiledFlexiKeys/.kiri
WEB_DIR_PATH = /mnt/d/Work/Projects/CustomMacropad/PCB/TiledFlexiKeys/.kiri/web/

| /home/rishabh/.local/share/kiri/bin/plot_kicad_sch: line 385: setxkbmap: command not found

    | Error when trying to load board at 'TiledFlexiKeys.kicad_pcb'
    | KiCad was unable to open this file because it was created with a more recent version than the one you are running.
    |
    | To open it you will need to upgrade KiCad to a version dated 01/08/24 or later.
    |
    | Full error text:
    | Unknown token 'generator_version' in 'TiledFlexiKeys.kicad_pcb', line 4, offset 3.
    | Wrong version of the API
    | Try sourcing 'env-nightly.sh' instead.
    |
    | Error when trying to load board at 'TiledFlexiKeys.kicad_pcb'
    | KiCad was unable to open this file because it was created with a more recent version than the one you are running.
    |
    | To open it you will need to upgrade KiCad to a version dated 10/18/22 or later.
    |
    | Full error text:
    | KiCad was unable to open this file because it was created with a more recent version than the one you are running.
    |
    | To open it you will need to upgrade KiCad to a version dated 10/18/22 or later.
    |
    | Full error text:
    | Expecting 'layer, width, fill, tstamp, locked, or status' in 'TiledFlexiKeys.kicad_pcb', line 571, offset 8.
    | Wrong version of the API
    | Try sourcing 'env-nightly.sh' instead.
    |
    | Exporting layouts only`

64BitDragon, 2024-03-29:
Not sure if it's the right solution but for me manually installing kicad 7 fixed this issue. It probably breaks something else but I don't know yet.

leoheck, 2024-03-29:
Yeah, Kicad 6 is kinda buggy.  Is it possible to share the repo of your design so I can check if I can run Kiri on it?

rishabhrkaushik, 2024-03-30:
I don't have the repo hosted yet. I'll send you the link once I have it pushed. 

Also, to be clear, my project is in Kicad 7 and later Kicad 8. I don't know why the software is using Kicad 6.

leoheck, 2024-04-02:
> I don't know why the software is using Kicad 6.

Which software are you talking about? Kiri doesn't use Kicad. Only if there is kicad-cli installed.

leoheck, 2024-04-02:
I still have to find time to verify what is going on. It would help if you could host a Kicad project for me to work on. Making a simple example to reproduce your issue is not that hard.

leoheck, 2024-04-02:
I pushed a fix to `kicadiff` so it will not fail when plotting Kicad 8 layouts. If this is the problem, it is fixed. Please, feel free to open this issue if the problem persists.

closed 2024-04-02 by leoheck