leoheck/kiri #114: Layout View is not showing my board

lowie2727, 2024-04-03:
when changing to to `Layout View` tab it keeps showing the schematic

kiri -v output

```zsh
    kiri 61f4b84 2024/04/02 20h26
    kicad 8.0.1-unknown-1000.20240330git1899058.fc39
    plotgitsch v0.9.0-8-ge5f94e4
    kidiff 9c4b17a 2024/04/02 20h25
```

shematic diff view works fine.

leoheck, 2024-04-04:
Dude common, help me here. how could you I fix this? there is zero information here to help me.
Can you at leat run it with the `-D` flag to show me the command line output??

lowie2727, 2024-04-04:
Hi, I'm sorry but I commented on a closed issue. I'll post it again

This is the output i get when running `kiri -r -D -t 4`
---

```zsh
Using Kicad project ./microSD.kicad_pro

          KIRI_HOME_PATH = /home/lowie/.local/share/kiri
           KICAD_VERSION = 8.0.1-1.fc39
            CURRENT_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd
 KICAD_PRO_ABSOLUTE_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/microSD.kicad_pro
      REPO_ABSOLUTE_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd
  KICAD_PROJ_NESTED_PATH = .
KICAD_PROJ_ABSOLUTE_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
         OUTPUT_DIR_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri
            WEB_DIR_PATH = /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/web/

    # Rev local files
         project_path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
              project: microSD.kicad_pro
            schematic: microSD.kicad_sch
               layout: microSD.kicad_pcb
```
```zsh
Output folder: .kiri
Removing leftovers from the previous run
Get commits command
   git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | %ad | %an | %s' main -- microSD.kicad_sch microSD.kicad_sch microSD.kicad_pcb

Checking local changes
[ ] ./microSD.kicad_pcb
[ ] ./microSD.kicad_sch
Files have local changes? No

Commits list
     1	33bcf51    2024-04-03 16:11:32    lowie2727     changed copper finish + solder mask and silkscreen color update
     2	9b04313    2024-04-03 15:29:13    lowie2727     updated polygon
     3	b846bdb    2024-04-03 15:19:18    lowie2727     removed unused copper layers
     4	776ca78    2024-04-02 00:41:06    KatoWarson    silkscreen + BOM update
     5	81a828f    2024-04-02 00:27:30    lowie2727     silkscreen update
     6	da18d67    2024-03-14 01:15:11    lowie2727     minor fixes schematic and PCB
     7	537d29a    2024-03-07 13:42:15    lowie2727     added info to generate BOM file
     8	ac22bc5    2024-03-07 02:43:18    lowie2727     fully migrated to KiCad
     9	f8df56a    2024-03-06 13:57:44    lowie2727     added drawing sheet
    10	7ab7bdb    2024-03-05 20:37:33    lowie2727     Eagle import
    11	227a5f6    2024-03-05 20:22:20    lowie2727     blank KiCad project

Artifacts Summary
11 commits (55 comparisons)

Generating artifacts (may take some time)
```

```zsh
# 1/55 33bcf51 | 2024-04-03 16:11:32 | lowie2727 | changed copper finish + solder mask and silkscreen color update
# 2/55 9b04313 | 2024-04-03 15:29:13 | lowie2727 | updated polygon

    Retrieving commits:
    - Source path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
    - Output path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51

    Retrieving commits:
    - Source path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.
    - Output path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313

    # Rev 33bcf51 files
         project_path: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51
              project: microSD.kicad_pro
            schematic: microSD.kicad_sch
[... 158 lines cut ...]

    Sch: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/microSD.kicad_sch
    Pages: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/sch_sheets

    PCB: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/microSD.kicad_pcb
    Layers: /home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/pcb_layers

    # Plotting schematics, 33bcf51
    DIR: /home/lowie/git/solar cooker/kicad-adafruit-microsd
    kicad-cli sch export svg --black-and-white --no-background-color --output "/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51/_KIRI_/sch" "/home/lowie/git/solar \
    cooker/kicad-adafruit-microsd/.kiri/33bcf51/microSD.kicad_sch"
    | Plotted to '/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/33bcf51/_KIRI_/sch/microSD.svg'.
    | Done
    status: 0

    # Plotting schematics, 9b04313
    DIR: /home/lowie/git/solar cooker/kicad-adafruit-microsd
    kicad-cli sch export svg --black-and-white --no-background-color --output "/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/sch" "/home/lowie/git/solar \
    cooker/kicad-adafruit-microsd/.kiri/9b04313/microSD.kicad_sch"
    | Plotted to '/home/lowie/git/solar cooker/kicad-adafruit-microsd/.kiri/9b04313/_KIRI_/sch/microSD.svg'.
    | Done
    status: 0
```

It looks like when using the kicad-cli it doens't include the PCB files at all. It is only plotting schematics. Without the -k flag it looks like it is plotting the PCB without any errors but i just can't see it in the browser. It tried using a diffrent browser but no succes. Hope the outputs help a bit, let me know.

leoheck, 2024-04-04:
Please, lets talk on the other issue. I am going to close this one.

closed 2024-04-04 by leoheck