leoheck/kiri #115: Not showing all pages and no layout

ZinnerC, 2024-04-11:
I have a project which I'm comparing with several hierarchical sheets (5 total) but it only shows the first 3, missing out 4 and 5.

Also the PCB "layers" are not shown. if I toggle in between pages/layers I see some picture flickering up and disappearing.

Did I miss anything in the setup?

leoheck, 2024-04-11:
Hierarchical sheets are always a problem since Kicad is really shitty storing them. But let's see if we can make this work. The script that finds the hierarchy may not be working on your computer/os/project.

How are you running Kiri?
I suggest doing this:
```
kiri -r -D -t 4
```

-r to start fresh removing old runs since if they didn't work, they won't interfere this time.
-D to show commands Kiri run on command line, so we can see if something has failed 
-t 4 to limit the number of commits so it is faster to check if it works.

Can you share the output of the command line here?

Is is possible to share the repo so I can test this on my end too?

ZinnerC, 2024-04-15:
So I ran the command as you suggested and re executed the creation of the comparison.

The repository I can not share since it is a Company repository. But here is the output of the command line.

```
smyze@LAPTOP-CAZ:~/GitLab/pcb_development$ cd 21-000001\ -\ Proxy/
smyze@LAPTOP-CAZ:~/GitLab/pcb_development/21-000001 - Proxy$ kiri -r -D -t 4
Using Kicad project ./21-000001 - proxy.kicad_pro

          KIRI_HOME_PATH = /home/smyze/.local/share/kiri
           KICAD_VERSION = 7.0.11-7.0.11~ubuntu22.04.1
            CURRENT_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy
 KICAD_PRO_ABSOLUTE_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy/21-000001 - proxy.kicad_pro
      REPO_ABSOLUTE_PATH = /home/smyze/GitLab/pcb_development
  KICAD_PROJ_NESTED_PATH = 21-000001 - Proxy
KICAD_PROJ_ABSOLUTE_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy
         OUTPUT_DIR_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy/.kiri
            WEB_DIR_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy/.kiri/web/

    # Rev 21-000001 - Proxy files
         project_path: /home/smyze/GitLab/pcb_development/21-000001 - Proxy
              project: 21-000001 - proxy.kicad_pro
            schematic: 21-000001 - proxy.kicad_sch
      extra_sch_files: Raspberry Pi RP2040.kicad_sch
                       WIZnet W5100.kicad_sch
                       WIZnet W5100s.kicad_sch
                       i2c.kicad_sch
                       power.kicad_sch
               layout: 21-000001 - proxy.kicad_pcb

Output folder: .kiri
Removing leftovers from the previous run
Get commits command
   git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | %ad | %an | %s%d' main -- 21-000001 - proxy.kicad_sch 21-000001 - proxy.kicad_sch \
Raspberry Pi RP2040.kicad_sch WIZnet W5100.kicad_sch WIZnet W5100s.kicad_sch i2c.kicad_sch power.kicad_sch 21-000001 - proxy.kicad_pcb 1

Checking local changes
[ ] ./21-000001
[ ] ./-
[ ] ./proxy.kicad_pcb
[ ] ./21-000001
[ ] ./-
[ ] ./proxy.kicad_sch
[ ] ./Raspberry
[ ] ./Pi
[ ] ./RP2040.kicad_sch
[ ] ./WIZnet
[ ] ./W5100.kicad_sch
[ ] ./WIZnet
[ ] ./W5100s.kicad_sch
[ ] ./i2c.kicad_sch
[ ] ./power.kicad_sch
Files have local changes? No

Commits list
     1  9ad9934    2024-03-11 08:33:46    Cassian Zinner    revC ready for production
     2  6a78847    2024-03-09 00:33:47    Cassian Zinner    added TSV to HV I2C, optional PullUp for I2C, DNP some connectors
     3  8503fdb    2024-03-08 23:14:37    Cassian Zinner    corrected HV I2C (SDA/SCL switched)
     4  1d717c8    2024-03-08 22:29:56    Cassian Zinner    correction NPN transistor at 5Vi2c

Artifacts Summary
4 commits (6 comparisons)

Generating artifacts (may take some time)

-------------------------------------------------------------------------------------------------------------------------------------------------------------

# 1/6 9ad9934 | 2024-03-11 08:33:46 | Cassian Zinner | revC ready for production
# 2/6 6a78847 | 2024-03-09 00:33:47 | Cassian Zinner | added TSV to HV I2C, optional PullUp for I2C, DNP some connectors

[... 260 lines cut ...]
               layout: 21-000001 - proxy.kicad_pcb

Assembling the HTML
- Adding list of commits
- Adding list of pages
- Adding list of layers

Latest rev 9ad9934
- Project's title (folder name): pcb_development
- Project's name (file name): 21-000001 - proxy
- Sch title: 21-000001 - proxy
- PCB title: 21-000001 - proxy
- Sch revision: C
- PCB revision: C
- Sch date: 2024-03-11
- PCB date: 2024-03-11

Kiri main page
- Initial project path:
  Pro 1: .kiri/9ad9934/21-000001 - proxy.kicad_pro
  Pro 2: .kiri/6a78847/21-000001 - proxy.kicad_pro

Starting webserver at http://127.0.0.1:8080/web/index.html
(Hit Ctrl+C to exit)
```

leoheck, 2024-04-15:
I could spot 2 things by looking at the logs quickly.

Something weird happened here there are weird filenames. But this is not quite important.
```
Checking local changes
[ ] ./21-000001
[ ] ./-
[ ] ./proxy.kicad_pcb
[ ] ./21-000001
[ ] ./-
[ ] ./proxy.kicad_sch
[ ] ./Raspberry
[ ] ./Pi
[ ] ./RP2040.kicad_sch
[ ] ./WIZnet
[ ] ./W5100.kicad_sch
[ ] ./WIZnet
[ ] ./W5100s.kicad_sch
[ ] ./i2c.kicad_sch
[ ] ./power.kicad_sch
Files have local changes? No
```

And also this. Kidiff is not working as expected.
```
    # Plotting layouts 9ad9934, 8503fdb
    DIR: /home/smyze/GitLab/pcb_development/21-000001 - Proxy
    kidiff -k -n --webserver-disable -a 9ad9934 -b 8503fdb -o "/home/smyze/GitLab/pcb_development/21-000001 - Proxy/.kiri" -f "21-000001 - \
    proxy.kicad_pcb"
    | usage: plot_kicad_pcb.py [-h] [-o OUTPUT_FOLDER] [-q] [-l] [-v] [-f] [-n]
    |                          kicad_pcb
    | plot_kicad_pcb.py: error: unrecognized arguments: - proxy.kicad_pcb
    |
    | usage: plot_kicad_pcb.py [-h] [-o OUTPUT_FOLDER] [-q] [-l] [-v] [-f] [-n]
    |                          kicad_pcb
    | plot_kicad_pcb.py: error: unrecognized arguments: - proxy.kicad_pcb
    |
    | Exporting layouts only
    |
    |       SCM Selected: git
    |    Kicad File Path: /home/smyze/GitLab/pcb_development/21-000001 - Proxy/21-000001 - proxy.kicad_pcb
    | Kicad Project Path: /home/smyze/GitLab/pcb_development/21-000001 - Proxy
    |          REPO Path: /home/smyze/GitLab/pcb_development
    |  Kicad Project Dir: 21-000001 - Proxy
    |    Board File Name: 21-000001 - proxy.kicad_pcb
    |         Output Dir: /home/smyze/GitLab/pcb_development/21-000001 - Proxy/.kiri
    |
    | Commit 1 (a): 9ad9934
    | Commit 2 (b): 8503fdb
    | Plotting the page with frame
    | Generate output files with layer id only
    | Error while plotting the layout with plot_kicad_pcb
    status: 1
```

Both issues may be due to problems with filenames. Why do you use spaces on the filenames? 
Do yourself a favor, don't use space on filenames for engineering things many programs don't like this. Also, it could be something related to the "hyphen" but I am not sure yet.

Kiri should work with spaces and hyphens, but it may have a bug related somewhere, I have to investigate this.

Also, your project is nested inside the GitHub repo. This is not normal too, it may introduce issues that I will have to check.

leoheck, 2024-04-15:
I pushed a fix for `Kicad-Diff`. Can you update your Kiri repo to see if it works?
Kicad diff was not handling spaces correctly, I am not sure if it is fixed tho. I could not test it yet since I don't have a repo with the same config that generates the problem.

leoheck, 2024-04-16:
hm it looks like it is not working yet.. I am going to fix this if I can.

leoheck, 2024-04-16:
This was hard. 

I think I got something working. It is pushed already. Please, could you test it and tell me if it is working for you?
It should plot the layout now, and also fix the list of files under the `Checking local changes`.

Now, the issue with the schematics is not fixed yet since I could not reproduce it here.
Could you describe how the hierarchy of your schematics is so I can try to make the same hierarchy as a test?

I am seeing these files.. but this does not tell me which is being used inside of each and maybe this is the big problem now.
```
       schematic: 21-000001 - proxy.kicad_sch
 extra_sch_files: Raspberry Pi RP2040.kicad_sch
                  WIZnet W5100.kicad_sch
                  WIZnet W5100s.kicad_sch
                  i2c.kicad_sch
                  power.kicad_sch
```

leoheck, 2024-04-16:
It seems that the issue with the schematic may have been resolved at least on my computer I could not reproduce it.

Additionally, I've managed to fix the layout plot functionality when using -k to use the kicad-cli. Though it runs slower, it now properly handles files with spaces in their names. 

I would appreciate your feedback to confirm whether these improvements are noticeable on your end.

jonathanbxxx, 2024-04-17:
I have a similar issue using KiCad 7.0.10 msys2 build. There is a bug in KiCad svg export. A page containing a net class directive label is creating a corrupted svg. 
[image]
[image]
The path on line 2837 is not correctly ended. It is missing the close brace.

leoheck, 2024-04-17:
hm, can you point me the right file of this? kiri script does not have more than 2640 lines

jonathanbxxx, 2024-04-17:
> [Tracking image][https://github.com/notifications/beacon/A64KMP2YDZ4WQOKYTKWMQHLY52SFXA5CNFSM6AAAAABGDC5IMSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT24PQCE.gif]

Sorry for the misunderstanding in the generated SVR file.

leoheck, 2024-04-17:
Please, create another issue to discuss this. Also, run kiri with `-D` and give me the command line output so I can understand the issue.

jonathanbxxx, 2024-04-17:
> What is the SRV file?

Sorry the svg file. It can be replicated by manually creating the svg file in the schematic editor. Then trying to open svg file in browser.

leoheck, 2024-04-17:
@ZinnerC, please reopen this report if you still see the issue.

leoheck, 2024-04-17:
@jonathanbxxx what the hell are you talking about? If you have any issue, please, open an issue report explaining what you are doing and what is not working.

closed 2024-04-17 by leoheck

ZinnerC, 2024-05-07:
@leoheck, so I gave it another try. Any it makes not difference to me as far I can judge.
first I did this: 
```
kiri --version
    kiri 3deff5e 2024/04/16 06h33
    kicad 7.0.11-7.0.11~ubuntu22.04.1
    plotgitsch v0.9.0-8-ge5f94e4
    kidiff eaf76ba 2024/04/16 06h33
```
So as I can see I'm up to date with the main branche.
Then I did: `kiri -r -D -t 4` in the same folder and I got the following output.

```
smyze@LAPTOP-CAZ:~/GitLab/pcb_development/21-000001 - Proxy$ kiri -r -D -t 4
Using Kicad project ./21-000001 - proxy.kicad_pro

          KIRI_HOME_PATH = /home/smyze/.local/share/kiri
           KICAD_VERSION = 7.0.11-7.0.11~ubuntu22.04.1
            CURRENT_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy
 KICAD_PRO_ABSOLUTE_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy/21-000001 - proxy.kicad_pro
      REPO_ABSOLUTE_PATH = /home/smyze/GitLab/pcb_development
  KICAD_PROJ_NESTED_PATH = 21-000001 - Proxy
KICAD_PROJ_ABSOLUTE_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy
         OUTPUT_DIR_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy/.kiri
            WEB_DIR_PATH = /home/smyze/GitLab/pcb_development/21-000001 - Proxy/.kiri/web/

    # Rev 21-000001 - Proxy files
         project_path: /home/smyze/GitLab/pcb_development/21-000001 - Proxy
              project: 21-000001 - proxy.kicad_pro
            schematic: 21-000001 - proxy.kicad_sch
      extra_sch_files: Raspberry Pi RP2040.kicad_sch
                       WIZnet W5100.kicad_sch
                       WIZnet W5100s.kicad_sch
                       i2c.kicad_sch
                       power.kicad_sch
               layout: 21-000001 - proxy.kicad_pcb

Output folder: .kiri
Removing leftovers from the previous run

Selecting commits
   git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | %ad | %an | %s%d' caz/21-000001_I2C-update -- \
   '21-000001 - proxy.kicad_sch' '21-000001 - proxy.kicad_sch' 'Raspberry Pi RP2040.kicad_sch' 'WIZnet W5100.kicad_sch' \
   'WIZnet W5100s.kicad_sch' 'i2c.kicad_sch' 'power.kicad_sch' '21-000001 - proxy.kicad_pcb' | cat

Checking local changes
[ ] ./power.kicad_sch
[ ] ./i2c.kicad_sch
[ ] ./WIZnet W5100s.kicad_sch
[ ] ./WIZnet W5100.kicad_sch
[ ] ./Raspberry Pi RP2040.kicad_sch
[ ] ./21-000001 - proxy.kicad_sch
[ ] ./21-000001 - proxy.kicad_pcb
Files have local changes? No

Commits list
     1  b197d64    2024-05-06 16:05:30    Cassian Zinner    update production files (HEAD -> caz/21-000001_I2C-update
     2  6d71411    2024-05-06 10:32:51    Cassian Zinner    REV D - updated: DEBUG, USB, 3v3 I2C, I2C PullUps
     3  9ad9934    2024-03-11 08:33:46    Cassian Zinner    revC ready for production
     4  6a78847    2024-03-09 00:33:47    Cassian Zinner    added TSV to HV I2C, optional PullUp for I2C, DNP some co

Artifacts Summary
4 commits (6 comparisons)

Generating artifacts (may take some time)

------------------------------------------------------------------------------------------------------------------------
# 1/6 b197d64 | 2024-05-06 16:05:30 | Cassian Zinner | update production files (HEAD -> caz/21-000001_I2C-updat
# 2/6 6d71411 | 2024-05-06 10:32:51 | Cassian Zinner | REV D - updated: DEBUG, USB, 3v3 I2C, I2C PullUps

[... 242 lines cut ...]
Latest rev b197d64
- Project's title (folder name): pcb_development
- Project's name (file name): 21-000001 - proxy
- Sch title: 21-000001 - proxy
- PCB title: 21-000001 - proxy
- Sch revision: D
- PCB revision: C
- Sch date: 2024-05-06
- PCB date: 2024-03-11

Kiri main page
- Initial project path:
  Pro 1: .kiri/b197d64/21-000001 - proxy.kicad_pro
  Pro 2: .kiri/6d71411/21-000001 - proxy.kicad_pro

Starting webserver at http://127.0.0.1:8080/web/index.html
(Hit Ctrl+C to exit)
```
It did not show all the layers. Something I found  which if think is strange the list of "plotted" files is not the same as you can find in the respective folder.

[image]

Also there are some statements about "Generate output files with layer id only" and "Drawing sheet ... not found". Could this be something?

Also the PCB diff does only flash up something while toggling between schematic and pcb.

ZinnerC, 2024-05-07:
@leoheck with regard to the "nested" repo. Well I wanted to do a repo per project. But I had several people complaining that it is to tideous to have multiple repositories to manage. Honestly, I would really like to have repositories per project. You don't want to see the monorepo for the firmwares. I'm thinking about to transfere it to individual repositories one point.
And in that go I could get rid of spaces and hyphens ;-)

ZinnerC, 2024-05-07:
@leoheck btw, I think I can not reopen that issue since I did not close it ;-)