leoheck/kiri #116: Unable to view schematic or layout diff (kiri: line 311: printf)

lpdx, 2024-05-15:
I have a similar issue like the https://github.com/leoheck/kiri/issues/107. Following the issue https://github.com/leoheck/kiri/issues/112 I've tried run with the options `-r -D -t 2 -k` flags but I got same message  `*kiri: line 311: printf*`.

My setup is: Ubuntu 22.04 in WSL 2 on Windows 11. 
The Kicad version are the same in windows and in the ubuntu.
And I've followed the [instalation tutorial](https://github.com/leoheck/kiri/blob/main/INSTALL.md) and all dependencies.

This is the log using -D flag:
```sh
          KIRI_HOME_PATH = /home/username/.local/share/kiri
           KICAD_VERSION = 7.0.11-7.0.11~ubuntu22.04.1
            CURRENT_PATH = /mnt/c/git/kicad/test
 KICAD_PRO_ABSOLUTE_PATH = /mnt/c/git/kicad/test/board.kicad_pro
      REPO_ABSOLUTE_PATH = /mnt/c/git/kicad/test
  KICAD_PROJ_NESTED_PATH = .
KICAD_PROJ_ABSOLUTE_PATH = /mnt/c/git/kicad/test/.
         OUTPUT_DIR_PATH = /mnt/c/git/kicad/test/.kiri
            WEB_DIR_PATH = /mnt/c/git/kicad/test/.kiri/web/

    # Rev local files
         project_path: /mnt/c/git/kicad/test/.
              project: board.kicad_pro
            schematic: board.kicad_sch
               layout: board.kicad_pcb

Selecting commits
   git log --date=format:'%Y-%m-%d %H:%M:%S' --pretty='format:%h | %ad | %an | %s%d' master -- 'board.kicad_sch' 'board.kicad_sch' \
   'board.kicad_pcb' | cat

Checking local changes
[ ] ./board.kicad_sch
[ ] ./board.kicad_pcb
Files have local changes? No

Commits list
     1  80ec49b    2024-05-15 13:23:28     username    add via (HEAD -> master)
     2  b553dcb    2024-05-14 15:02:21     username    schema changes
     3  3d848c1    2024-05-14 11:39:04     username    initial commit

Artifacts Summary
3 commits (3 comparisons)

Generating artifacts (may take some time)
    # Rev local files
         project_path: /mnt/c/git/kicad/test/.
              project: board.kicad_pro
            schematic: board.kicad_sch
               layout: board.kicad_pcb

Assembling the HTML
- Adding list of commits
- Adding list of pages
- Adding list of layers
/home/username/.local/share/kiri/bin/kiri: line 311: printf: at: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: attr: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: clearance: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: connect_pads: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: creategerberjobfile: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: dashed_line_dash_ratio: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: dashed_line_gap_ratio: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: descr: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: dimension: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: disableapertmacros: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: drillshape: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: dxfimperialunits: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: dxfpolygonmode: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: dxfusepcbnewfont: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: effects: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: fill: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: filled_polygon: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: footprint: invalid number
[... 42 lines cut ...]
/home/username/.local/share/kiri/bin/kiri: line 311: printf: segment: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: setup: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: sketchpadsonfab: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: stroke: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: style: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: subtractmaskfromsilk: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: svgprecision: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: tags: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: tstamp: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: useauxorigin: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: usegerberadvancedattributes: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: usegerberattributes: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: usegerberextensions: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: via: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: viasonmask: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: xy: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: zone: invalid number
/home/username/.local/share/kiri/bin/kiri: line 311: printf: zone_connect: invalid number
```

When I start the http server in Windows (via `python -m http.server 3306` ) I can see the board but not the changes:
[image]

Another point (maybe because of the issue above) there are some strange data in the "Layers" section of the html page:
[image]

leoheck, 2024-05-15:
Hi @lpdx do you have printf installed? 

Did you run this script to install dependencies?
```
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"
```

leoheck, 2024-05-15:
The layers list is not being created correctly for some reason, for you. And this is the issue.

leoheck, 2024-05-15:
I think a program is missing, or maybe there is a program that has different settings on your machine. Do you have a custom $PATH? Can you show me the content of this variable?  You can get it with this command `echo $PATH`

lukas-lang, 2024-05-27:
I encounter the same error, even when using the kiri-docker container. So assuming that one is set up correctly, it's not an issue with the environment?

lpdx, 2024-05-27:
> Hi @lpdx do you have printf installed?
> 
> Did you run this script to install dependencies?
> 
> ```
> bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"
> ```
 Yes, I've used this script to install the dependencies.

> I think a program is missing, or maybe there is a program that has different settings on your machine. Do you have a custom $PATH? Can you show me the content of this variable? You can get it with this command echo $PATH

username@computer: echo $PATH
/home/username/.local/share/blender:/home/username/.local/share/kiri/bin:/home/username/.local/share/kiri/submodules/KiCad-Diff/bin:/home/username/.opam/4.10.2/bin:/home/username/.vscode-server/bin/dc96b837cf6bb4af9cd736aa3af08cf8279f7685/bin/remote-cli:/home/username/.local/bin:/home/username/bin:/home/username/.local/share/blender:/home/username/.local/share/kiri/bin:/home/username/.local/share/kiri/submodules/KiCad-Diff/bin:/home/username/.opam/4.10.2/bin:/home/username/miniconda/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files (x86)/Common Files/Oracle/Java/java8path:/mnt/c/Program Files (x86)/Common Files/Oracle/Java/javapath:/mnt/c/Program Files/Eclipse Adoptium/jre-21.0.2.13-hotspot/bin:/mnt/c/Program Files/Typhoon HIL Control Center 2024.1/bin/:/mnt/c/windows/system32:/mnt/c/windows:/mnt/c/windows/System32/Wbem:/mnt/c/windows/System32/WindowsPowerShell/v1.0/:/mnt/c/windows/System32/OpenSSH/:/mnt/c/Program Files/Git/cmd:/mnt/c/Users/username/mingw32/bin/:/mnt/c/Users/username/git/utils/git-filter-repo/:/mnt/c/Program Files/MiKTeX/miktex/bin/x64:/mnt/c/Program Files/LLVM/bin:/mnt/c/Program Files/doxygen/bin:/mnt/c/Program Files/Graphviz/bin:/mnt/c/ProgramData/miniconda3:/mnt/c/ProgramData/miniconda3/Scripts:/mnt/c/ProgramData/miniconda3/Library/bin:/mnt/c/msys64/mingw64/bin:/mnt/c/cygwin64/bin:/mnt/c/Users/username/.local/share/gem/ruby/3.2.0/gems/ceedling-0.31.1/bin:/mnt/c/Users/username/.local/share/gem/ruby/3.2.0/bin:/mnt/c/ProgramData/chocolatey/bin:/mnt/c/Program Files/PuTTY/:/mnt/c/Ruby32-x64/bin:/mnt/c/Users/username/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/username/AppData/Local/GitHubDesktop/bin:/mnt/c/Users/username/AppData/Local/Programs/Microsoft VS Code/bin:/mnt/c/Users/username/AppData/Local/Programs/MiKTeX/miktex/bin/x64/:/snap/binFiles/doxygen/bin:/mnt/c/Program Files/Graphviz/bin:/mnt/c/ProgramData/miniconda3:/mnt/c/ProgramData/miniconda3/Scripts:/mnt/c/ProgramData/miniconda3/Library/bin:/mnt/c/msys64/mingw64/bin:/mnt/c/cygwin64/bin:/mnt/c/Users/andre.leopoldino/.local/share/gem/ruby/3.2.0/gems/ceedling-0.31.1/bin:/mnt/c/Users/andre.leopoldino/.local/share/gem/ruby/3.2.0/bin:/mnt/c/ProgramData/chocolatey/bin:/mnt/c/Program Files/PuTTY/:/mnt/c/Ruby32-x64/bin:/mnt/c/Users/andre.leopoldino/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/andre.leopoldino/AppData/Local/GitHubDesktop/bin:/mnt/c/Users/andre.leopoldino/AppData/Local/Programs/Microsoft VS Code/bin:/mnt/c/Users/andre.leopoldino/AppData/Local/Programs/MiKTeX/miktex/bin/x64/:/snap/bin

leoheck, 2024-05-27:
Kiri-docker may be a bit outdated. Did you try the last updates @lukas-lang ?

lukas-lang, 2024-05-27:
@leoheck Yes, I rebuilt the docker container manually now, and the issue persists. `sudo cat kiri_version.json` within the container gives
```
{"ref":"refs/heads/main","node_id":"MDM6UmVmMzQ3Nzc2OTE4OnJlZnMvaGVhZHMvbWFpbg==","url":"https://api.github.com/repos/leoheck/kiri/git/refs/heads/main","object":{"sha":"b4a1e2818f3ef3d45743a6e747e8942b560d73b0","type":"commit","url":"https://api.github.com/repos/leoheck/kiri/git/commits/b4a1e2818f3ef3d45743a6e747e8942b560d73b0"}}
```
which seems to be the latest commit

lukas-lang, 2024-05-27:
It looks like the issue is caused by the Windows line endings in the `_local_` version: While all the actual commits are fine (since line endings are checked out UNIX style), the "live" repository version is not (it contains Windows line endings), which breaks the parsing of the PCB. Adding another pre-processing step to `parse_layout_layers` seems to fix the issue (at least for this, there might be other places where the line endings cause issues)
```bash
	if [[ ${pcb_version} -ge 20240108 ]]; then   # (version 20240108) # Kicad 8

		local layers=$( \
			tr -d '\r' < "${kicad_pcb}" |
			sed -n '/^\t(layers/,/\t)$/{/^\t(layers/!{/\t)$/!p}}' | \
			sed "s/^[ \t]\+//g" | \
			sed "s/[()]//g" | \
			sed "s/\r$//g" | \
```

leoheck, 2024-05-27:
Ah, that's a good help, thanks. I was struggling to get a windows to verify this. But I think I know how do fix that.

lukas-lang, 2024-05-28:
Thanks for the quick fix! It looks like it almost works: I needed to change line 1406 as follows to prevent `sed` from erroring out on the subdirectories (such as `_KIRI_`)

    sed -i 's/\r//' "${OUTPUT_DIR_PATH}/_local_/"*.* &> /dev/null

closed 2024-05-28 by leoheck

leoheck, 2024-05-30:
Hey @lpdx 

Your log shows this `sed: preserving permissions for`.
This may indicate that you don't have permissions to do stuff on that folder for some reason.

Please, check the following link to see if this fixes the issue for you.

https://stackoverflow.com/questions/14313318/permission-denied-when-sed-in-place-edit-in-mingw#14410957

leoheck, 2024-06-02:
Also, you shared a sample project @lpdx but you have to share a project inside a git repo, or a url to the project on github for instance otherwise kiri wont work.

leoheck, 2024-06-02:
> `sed: preserving permissions for ‘/mnt/c/git/test/.kiri/web/sedmyj4kK’: Operation not permitted`

Ah, I see what you are trying to do... you are mounting a network drive or Windows drive to access your files from within the WLS but you don't have permissions or it is read-only, or maybe your terminal does not have permissions to access files on Windows partitions.