leoheck/kiri #117: root index.html redirects to absolute path

milesvp, 2024-05-16:
There is a minor "bug" in index.html that redirects to /web/index.html rather than web/index.html. This would allow a relative path and would allow people to host in a subdirectory of a webserver. This is a use case I tend to use to collaborate.

https://example.com/somedir/mykicadproject/  ->  https://example.com/somedir/mykicadproject/web/index.html

best as I can tell there is only a one line diff, and it works for the redirect. 

desktop:~/dev/kiri$ git diff
diff --git a/assets/redirect.html b/assets/redirect.html
index 722adbe..a2ee85c 100644
--- a/assets/redirect.html
+++ b/assets/redirect.html
@@ -1,7 +1,7 @@
 <!DOCTYPE html>
 <html>
 <head>
-  <meta http-equiv='refresh' content='0; URL=/web/index.html'>
+  <meta http-equiv='refresh' content='0; URL=web/index.html'>
 </head>
 <body>
 </body>

leoheck, 2024-05-20:
But how? Does the number of backslashes matter?

I am doing this.. and this works fine.
````
http://127.0.0.1:8080/web///////////index.html
```

It is like Linux paths, the number of `/` does not matter at all.

milesvp, 2024-05-21:
it's the first slash that matters. same as the linux cli. / is root. so /web always forwards to /web, but web forwards to some_relative_path/web.

To test this you'd need to put the files in another location. 

http://127.0.0.1:8080/somesubdir/ will currently redirect to http://127.0.0.1:8080/web/index.html

then you'll get a 404 because the files all live in somesubdir/

leoheck, 2024-05-21:
I did not know if I understood the problem well, I guess.

However, I found this, to guide me through this issue.
https://webmasters.stackexchange.com/questions/56840/what-is-the-purpose-of-leading-slash-in-html-urls

So, it looks like you want a more flexible way to handle this redirect. And since I could not see any direct issue of doing this, I made the change. It is pushed on the repo already. 

Please, check if this improves something for you, and report back here if needed. I am closing this issue for now. Cool?

closed 2024-05-21 by leoheck