leoheck/kiri #119: PCB cutoff when past A4 page borders

milesvp, 2024-08-03:
Apologies. This looks like a problem with kidiff, but there is no Issue tracker at https://github.com/leoheck/KiCad-Diff.

When I try to use kiri with a PCB that is outside of the default A4 borders. The resulting SVG files are cut off, looks like they cut off at the page border. I see no way to get around this. I've tried poking around with kidiff, but I see no way at the command line for it to ensure that it's got all the data.

Further apologies if this is a kicad plotting bug.

I also understand if this is a "won't fix". In the future my partner and I will avoid commiting PCBs that are outside the page bounderies. But would be a nice fix for one of our existing boards.

leoheck, 2024-08-03:
No need to apologize. That's something wierd with Kicad or the SVG plots of Kicad. 

Did you try to open the generated SVG files manually with another program to check if the information exists there? I would suggest using inkscape.

If the information is there, but it is not being displayed correctly in the browser, it is just a matter of optimizing the SVG before of displaying it with kiri. You can do this tweaking kiri (I guess I already have something there tweaking svgs, if you have the tools installed). Another alternative is to run kiri, without launching the server, the annually tweaking the svgs with an script and then running Kiri again without removing previous files.

Ah, thinking here, another solution would be to change the page size before of plotting... from A4 to A3 for instance. This solution could work. 

An extra issue here is that if the content is above, on the left or on the top-left of the page, this may be a problem and it may not be presented at all considering some of my tests in the past.

leoheck, 2024-08-03:
If you can share your project I can also give it a look from my side.

leoheck, 2024-08-03:
I have an old project that I used to test this in the past.
This is working for both schematics and layouts on my computer.

See:

[image]

[image]

When you run kiri with `-r`, do you see this `Optimizing svg for web` step?

[image]

This step works only when you have the `svgo` installed and available on the PATH. It is not on the dependencies script since it is not kind of a dependency. If you install it, it may fix your problem.

milesvp, 2024-08-04:
Afraid I can't share the project, as it's client work. But here's the output I'm seeing. 

This is in what I believe is the latest kiri-docker

kiri -r -e "FR4\|MetalCore" -D PulseControlACBoard/PulseControlACBoard.kicad_pro

[image]

[image]

This is the PulseControlACBoard/.kiri/bbf292f/_KIRI_/pcb/layer-00.svg
[image]

# Plotting layouts bbf292f, fe0a836
    DIR: /home/kiri/hardware/PulseControlACBoard
    kidiff -k -n --webserver-disable -a bbf292f -b fe0a836 -o "/home/kiri/hardware/PulseControlACBoard/.kiri" -f "PulseControlACBoard.kicad_pcb"
    | 06:19:06: Warning: Font 'Bitstream Vera Sans Mono' not found; substituting 'DejaVu Sans Mono Bold'.
    | 06:19:06: Warning: Font 'Bitstream Vera Sans' not found; substituting 'DejaVu Sans Bold'.
    | 06:19:06: Warning: Font 'Noto Sans Display ExtraBold' not found; substituting 'DejaVu Sans Book'.
    | 
    | Exporting layouts only
    | 
    |       SCM Selected: git 
    |    Kicad File Path: /home/kiri/hardware/PulseControlACBoard/PulseControlACBoard.kicad_pcb
    | Kicad Project Path: /home/kiri/hardware/PulseControlACBoard
    |          REPO Path: /home/kiri/hardware
    |  Kicad Project Dir: PulseControlACBoard
    |    Board File Name: PulseControlACBoard.kicad_pcb
    |         Output Dir: /home/kiri/hardware/PulseControlACBoard/.kiri
    | 
    | Commit 1 (a): bbf292f
    | Commit 2 (b): fe0a836
    | Plotting the page with frame
    | Generate output files with layer id only
    Status: 0
    --

    # Rev PulseControlACBoard files
         project_path: /home/kiri/hardware/Pulse

I just tried to install npm, to install svgo on the docker container. Yikes. It's a lot of files. I can't tell if your example is just short enough to not run into the problem I'm seeing, but I'll try it on a clean container tomorrow.