leoheck/kiri #121: Kiri button in pcb editor fails to launch kiri

jp-dt, 2024-11-01:
I've successfully installed kiri and can run it from the command line (kiri path/to/project.kicad_pro), but when I try to launch it from the plugin button in the pcb editor it fails with the following exception:

```
Traceback (most recent call last):

  File "/home/username/.local/share/kicad/8.0/scripting/plugins/kiri/KiRI.py", line 59, in Run
    process = subprocess.Popen(cmd, stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE, **kwargs)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,

  File "/usr/lib/python3.12/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)

FileNotFoundError: [Errno 2] No such file or directory: 'kiri'
```

Other info:
OS: Ubuntu 24.04.1
KiCad: 8.0.6
Kiri: commit 18199e1

leoheck, 2024-11-05:
Can you double check if you have the right PATH set?
Here on my machine it is working.

Alternatively, if you need a workaround, you could edit this file:
```python
/home/username/.local/share/kicad/8.0/scripting/plugins/kiri/KiRI.py
``` 

Updating `kiri` on lines 45 and 47 by the whole kiri path.
Like:

```python
            cmd = ['/path/to/kiri/bin/kiri', project_file_path, "-u", "-t", "2", "-r"]
        except:
            cmd = ['/path/to/kiri/bin/kiri', "-u", "-t", "2", "-r"]
```

leoheck, 2024-11-06:
I improved Kiri and the Kiri plugin regarding this issue, but I still think Kiri is not in the PATH.
Can you update your cloned repo to check this?

Can you share the result of these commands (running them on a terminal)?
```bash
echo $HOME
which kiri
```

Also it looks like your path is a bit off but I am not sure if this is the issue. You have a path starting with `/home/username/` which tells me the username is `username` is this correct? If not, you should update the `username` field with the real username and it may work.

jp-dt, 2024-11-06:
I anonymized my username before posting, old habit.

I re-installed Kiri by re-running the command 

```
bash -c "INSTALL_KIRI_REMOTELLY=1; \
    $(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_kiri.sh)"
```

but there was no change. Trying to press the button to run the plugin still gives the same exception as in my initial post.

I tried re-running the script to install the plugin, by running the command 

```
bash -c "INSTALL_KIRI_REMOTELLY=1; \
    $(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_plugin.sh)"
```

and got the following output:

```
Traceback (most recent call last):
  File "/home/jpotts/.local/share/kiri/bin/kicad_version.py", line 23, in <module>
    import pcbnew as pn
ModuleNotFoundError: No module named 'pcbnew'
Kiri plugin installed in /home/jpotts/.local/share/kicad//scripting/plugins/kiri
```

This is a bit odd, since pcbnew is in the path and runs fine.

Here's the result of the commands you asked for:

```
jpotts@ENG-44-LX:~$ echo $HOME
/home/jpotts
jpotts@ENG-44-LX:~$ which kiri
/home/jpotts/.local/share/kiri/bin/kiri
```

I'm able to run kiri directly, with the following command/output:

```
kiri ~/Documents/kicad_repos/3080/3080.kicad_pro

Artifacts Summary
5 commits (10 comparisons)

Generating artifacts (may take some time)

Assembling the HTML
- Adding list of commits
- Adding list of pages
- Adding list of layers

Latest rev f5c3b4b
- Project's title (folder name): 3080
- Project's name (file name): 3080
- Sch title: 3080
- PCB title: 3080
- Sch revision: 00
- PCB revision: 00
- Sch date: 2024-11-05
- PCB date: 2024-11-05

Kiri main page
- Initial project path:
  Pro 1: .kiri/f5c3b4b/3080.kicad_pro
  Pro 2: .kiri/3be8ebf/3080.kicad_pro
Traceback (most recent call last):
  File "/home/jpotts/.local/share/kiri/bin/kicad_version.py", line 23, in <module>
    import pcbnew as pn
ModuleNotFoundError: No module named 'pcbnew'

Starting webserver at http://127.0.0.1:8080/web/index.html
(Hit Ctrl+C to exit)
```

Again there's the strange ModuleNotFoundError exception (I don't recall if that was popping up when I ran it before or not), but the program runs fine.

I did some further searching on that error, apparently it's an [been an issue](https://forum.kicad.info/t/kicad-8-on-ubuntu-modulenotfounderror-no-module-named-pcbnew/49029) but should be fixed. I added the following to .bashrc as a workaround:

```
export PYTHONPATH=/usr/lib/python3/dist-packages/
```

and then re-ran the plugin install script:

```
jpotts@ENG-44-LX:~$ bash -c "INSTALL_KIRI_REMOTELLY=1; \
    $(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_plugin.sh)"
Kiri plugin installed in /home/jpotts/.local/share/kicad/8.0/scripting/plugins/kiri
```

(seems to be a different path than it was installing to previously) Now when I click the action plugin button in the pcb editor, nothing happens. No exception is raised, but kiri also does not run.

jp-dt, 2024-11-06:
For good measure, I tried replacing "kiri" with the path to kiri in the plugin script located at ~/.local/share/kicad/8.0/scripting/plugins/kiri. After your changes, it looks like that's done with the kiri_bin variable on line 64, which I changed from 
```
kiri_bin = "kiri"
```
to
```
kiri_bin = "~/.local/share/kiri/bin/kiri"
```

however, this didn't make any difference.

leoheck, 2024-11-06:
Did you install Kiri dependencies? 

This is a missing dependency.
```python
ModuleNotFoundError: No module named 'pcbnew
```

leoheck, 2024-11-06:
This script here
```
bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"
```

jp-dt, 2024-11-06:
Yes, I installed the dependencies before installing kiri. To doublecheck, I re-ran that script just now, no change. I also re-ran the install script after, still no change.

As I mentioned, pcbnew not being found is a something of a known issue. Based on the discussion at the link I posted, it's related to sysconfig being used instead of distutils in python 3.12, and pcbnew not being exactly where sysconfig expects it to be. To address this, I ran `find / -name pcbnew.py` and found that it's located at `/usr/lib/python3/dist-packages/pcbnew/py`. So, I added `export PYTHONPATH=/usr/lib/python3/dist-packages/` to my .bashrc, and now everything can find pcbnew.

jp-dt, 2024-11-06:
Instead of launching kicad and opening the pcb from there, I ran pcbnew from command line to get the output. When I click the kiri button, the command line prints out the following:

```
jpotts@ENG-44-LX:~$ pcbnew Documents/kicad_repos/3080/3080.kicad_pcb
Starting KiRI plugin...
======================================
kiri -t 2 -r -R -V -l /home/jpotts/Documents/kicad_repos/3080/3080.kicad_pro
pid=9196

          KIRI_HOME_PATH = /home/jpotts/.local/share/kiri
           KICAD_VERSION = 8.0.6-8.0.6-0~ubuntu24.04.1
            CURRENT_PATH = /home/jpotts
 KICAD_PRO_ABSOLUTE_PATH = /home/jpotts/Documents/kicad_repos/3080/3080.kicad...
      REPO_ABSOLUTE_PATH = /home/jpotts/Documents/kicad_repos/3080
  KICAD_PROJ_NESTED_PATH = .
KICAD_PROJ_ABSOLUTE_PATH = /home/jpotts/Documents/kicad_repos/3080/.
         OUTPUT_DIR_PATH = /home/jpotts/Documents/kicad_repos/3080/.kiri
            WEB_DIR_PATH = /home/jpotts/Documents/kicad_repos/3080/.kiri/web/...

    # Rev local files
         project_path: /home/jpotts/Documents/kicad_repos/3080/.
              project: 3080.kicad_pro
            schematic: 3080.kicad_sch
      extra_sch_files: gate_driver.kicad_sch
                       phase.kicad_sch
               layout: 3080.kicad_pcb

Output folder: .kiri
Removing leftovers from the previous run

Commits list
     1	f5c3b4b    2024-11-05 18:44:54    John Potts    changed to 2oz outer 
     2	3be8ebf    2024-11-05 15:40:17    John Potts    milled between ESC an

Artifacts Summary
2 commits (1 comparisons)

Generating artifacts (may take some time)

--------------------------------------------------------------------------------

# 1/1 f5c3b4b | 2024-11-05 18:44:54 | John Potts | changed to 2oz outer
# 2/1 3be8ebf | 2024-11-05 15:40:17 | John Potts | milled between ESC a

    # Rev f5c3b4b files
         project_path: /home/jpotts/Documents/kicad_repos/3080/.kiri/f5c3b4b
              project: 3080.kicad_pro
            schematic: 3080.kicad_sch
      extra_sch_files: gate_driver.kicad_sch
                       phase.kicad_sch
               layout: 3080.kicad_pcb

    # Rev f5c3b4b project info
        Kicad Version: 8
          Sch Version: 20231120
          PCB Version: 20240108

    # Rev 3be8ebf files
         project_path: /home/jpotts/Documents/kicad_repos/3080/.kiri/3be8ebf
              project: 3080.kicad_pro
            schematic: 3080.kicad_sch
      extra_sch_files: gate_driver.kicad_sch
                       phase.kicad_sch
               layout: 3080.kicad_pcb

    # Rev 3be8ebf project info
        Kicad Version: 8
          Sch Version: 20231120
          PCB Version: 20240108

    # Plotting schematics, f5c3b4b
    DIR: /home/jpotts/Documents/kicad_repos/3080
    kicad-cli sch export svg --black-and-white --no-background-color \
    --output "/home/jpotts/Documents/kicad_repos/3080/.kiri/f5c3b4b/_KIRI_/sch" \
    "/home/jpotts/Documents/kicad_repos/3080/.kiri/f5c3b4b/3080.kicad_sch"
    Status: 0
    --

    # Plotting schematics, 3be8ebf
    DIR: /home/jpotts/Documents/kicad_repos/3080
    kicad-cli sch export svg --black-and-white --no-background-color \
    --output "/home/jpotts/Documents/kicad_repos/3080/.kiri/3be8ebf/_KIRI_/sch" \
    "/home/jpotts/Documents/kicad_repos/3080/.kiri/3be8ebf/3080.kicad_sch"
    Status: 0
    --

    # Plotting layouts f5c3b4b, 3be8ebf
    DIR: /home/jpotts/Documents/kicad_repos/3080
    kidiff -k -n --webserver-disable -a f5c3b4b -b 3be8ebf -o \
    "/home/jpotts/Documents/kicad_repos/3080/.kiri" -f "3080.kicad_pcb"
    Status: 0
    --

    # Rev local files
         project_path: /home/jpotts/Documents/kicad_repos/3080/.
              project: 3080.kicad_pro
            schematic: 3080.kicad_sch
      extra_sch_files: gate_driver.kicad_sch
                       phase.kicad_sch
               layout: 3080.kicad_pcb

Assembling the HTML
- Adding list of commits
- Adding list of pages
- Adding list of layers

Latest rev f5c3b4b
- Project's title (folder name): 3080
- Project's name (file name): 3080
- Sch title: 3080
- PCB title: 3080
- Sch revision: 00
- PCB revision: 00
- Sch date: 2024-11-05
- PCB date: 2024-11-05

Kiri main page
- Initial project path:
  Pro 1: .kiri/f5c3b4b/3080.kicad_pro
  Pro 2: .kiri/3be8ebf/3080.kicad_pro

Launching kiri-server (may take some time)...
kiri -t 2 -r -R -u &
pid=10738

DONE...
```

However, the webpage doesn't seem to launch. It doesn't just not open, if I navigate to http://127.0.0.1:8080/web/index.html, I get "The site cannot be reached".

As mentioned previously, I can run kiri directly from the command line with `kiri path/to/project.kicad_pro`, so this isn't critical.

leoheck, 2024-11-06:
wait a bit more... 
the last command `kiri -t 2 -r -R -u &` takes some time to run..
I will try to replace this with something faster...

leoheck, 2024-11-06:
Now I am calling kiri-server directly, it may take less time to launch. Hope it works.

leoheck, 2024-11-07:
It looks like you are almost there.
I recomend you to upgrade your repo once more since I replaced the last kiri comand with kiri-server...
Then as for the first use... do this:

Launch kicad from the command line, this way:

```
kicad project.pro
``` 

Then go to your board layout.
There use the KiRI plugin.
And check the command line to see kiri's progress there.
If everything goes well, It should run fine and at the end it should run kiri-server and launch the browser with the results.

leoheck, 2024-11-07:
I am closing this issue since it looks like it is solved. But if the problem still persists, please feel free to continue the conversation here.

jp-dt, 2024-11-07:
I ran the script to update, ran kicad with `kicad path/to/project.kicad_pro`, then opened the pcb editor and clicked the button for kiri. It ran through and generated the diffs etc, ending with the following:

```
Launching kiri-server (may take some time)...
kiri-server .kiri
pid=12784

DONE...
```

However, nothing happened.

I opened a new command line, entered `ps aux`, and looked for pid 12784, and saw the following: 
```
jpotts     12784  0.1  0.0      0     0 ?        Zs   12:27   0:00 [python3] <defunct>
```

ie kiri-server is dying silently and sitting there as a zombie process.

jp-dt, 2024-11-07:
Actually, my fix only works if I've run kicad from the command line. If I start kicad normally, I still have nothing happen when I click the kiri action plugin button in the pcb editor.

leoheck, 2024-11-08:
hm.. ok, so when launching kicad from GUI it does not go to the path of kicad project. and you are also not running it from the project folder on the command line. Then this makes sense. I have to check how to improve kiri plugin then.. also, kiri output should be given to the GUI so the user wont be lost during the process.