leoheck/kiri #126: Arguments for --older and --newer are interchanged

semiversus, 2025-01-28:
Here is an example of commits:
```
9a2467e 2023-1-1 The initial commit
7b349a1 2024-1-1 A simple change
91cd12d 2025-1-1 The latest change
```

When I use `--older 7b349a1` I get the commits 7b349a1 (2024) and  91cd12d (2025). But *older* should be 7b349a1 (2024) and 9a2467e (2023). The same, but reversed should work on `--newer`.

leoheck, 2025-01-29:
I have to check this but thinking on this a bit, if I am not mistaking, by `--older` I meant to use commits until the older commit is found. The same applies to the `--newer`. Also, checking the source quickly on my smartphone (not a good place for that) could see that maybe the help message is switched if I am not mistaking.

This is the current message:
```
-o|--older HASH  Show commits starting from this one. It uses the short hash.
-n|--newer HASH  Show commits until this one. It uses the short hash.
```

And this is what it should be:
```
-o|--older HASH  Show commits until this one. It uses the short hash.
-n|--newer HASH  Show commits starting from this one. It uses the short hash.
```

Does it make sense for you?

semiversus, 2025-01-29:
Currently, the help text is describing what is happening. As you wrote "--older" should get the commits until the older one is found (starting at the initial commit). The behavior is switched in the current implementation. Happy to support you with a pull request, if you can confirm this bug (may be I'm just wrong). Cool tool by the way.

leoheck, 2025-01-29:
I intended to limit the list of commits.

By default, if you don't use these flags, kiri processes the whole list of commits (that have changes on sch/kicad_sch/kicad_pcb) starting from the latest commits to the oldest.

Testing this I saw that the commands `kiri ls` and `kiri lsa` were broken.
These command lists commits Kiri is going to process without processing them.

For instance, I have this commit list on a project (in this case commits with changes only)
```bash
➜ kiri ls
     1	a7bb19c    2024-04-02 15:07:25    Leandro Heck    Update to Kicad 8 (HEAD -> main, origin/main, origin/HEAD)
     2	7aebd2e    2023-03-22 16:09:20    Leandro Heck    Update to Kicad 7
     3	5be11dd    2021-12-29 14:58:15    Leandro Heck    Update 3d models dir variable (selected_layers)
     4	d0628e0    2021-12-29 14:54:10    Leandro Heck    Convert project to Kicad v6
     5	010b611    2021-12-29 14:25:14    Leandro Heck    Update using kicad 5.1.12 (tag: 1.0)
     6	20e163c    2021-02-09 10:53:38    Leandro Heck    Remove 3Ds, fix 3V3 and 1V8 switched labels
     7	f35d811    2021-01-23 11:23:21    Leandro Heck    Align filled zones on 0.5 mm grid
     8	0a59275    2021-01-23 11:14:10    Leandro Heck    Update copper areas
     9	e91e087    2021-01-22 18:28:56    Leandro Heck    Final touches with the labels
    10	89f22a1    2021-01-22 17:55:51    Leandro Heck    Update layout file
    11	b59502b    2021-01-22 17:52:07    Leandro Heck    Update
    12	224bb02    2021-01-22 17:21:39    Leandro Heck    Initial version
```

This is the expected result when using `-n` (newer)  flag:
This command is telling to use the commits (from the beginning of the history until the `e91e087`, which is item #9 on the list above)
```bash
➜ kiri ls -n e91e087
     1	e91e087    2021-01-22 18:28:56    Leandro Heck    Final touches with the labels
     2	89f22a1    2021-01-22 17:55:51    Leandro Heck    Update layout file
     3	b59502b    2021-01-22 17:52:07    Leandro Heck    Update
     4	224bb02    2021-01-22 17:21:39    Leandro Heck    Initial version
```

Alternatively, the flag `-o` does the opposite, it starts from commit e91e087 (commit #9) until the newest one.
```bash
➜ kiri ls -o e91e087
     1	a7bb19c    2024-04-02 15:07:25    Leandro Heck    Update to Kicad 8 (HEAD -> main, origin/main, origin/HEAD)
     2	7aebd2e    2023-03-22 16:09:20    Leandro Heck    Update to Kicad 7
     3	5be11dd    2021-12-29 14:58:15    Leandro Heck    Update 3d models dir variable (selected_layers)
     4	d0628e0    2021-12-29 14:54:10    Leandro Heck    Convert project to Kicad v6
     5	010b611    2021-12-29 14:25:14    Leandro Heck    Update using kicad 5.1.12 (tag: 1.0)
     6	20e163c    2021-02-09 10:53:38    Leandro Heck    Remove 3Ds, fix 3V3 and 1V8 switched labels
     7	f35d811    2021-01-23 11:23:21    Leandro Heck    Align filled zones on 0.5 mm grid
     8	0a59275    2021-01-23 11:14:10    Leandro Heck    Update copper areas
     9	e91e087    2021-01-22 18:28:56    Leandro Heck    Final touches with the labels
```

They can be used together to get something like this or with the `-t` to limit the number of processed commits.

The changes were pushed fixing `ls` and `lsa` commands and fixing the issues with the help message of those commands.

semiversus, 2025-01-29:
Thanks for digging deeper. Filtering using `--newer` and `--older` is a very nice feature for projects with many commits.

The initial problem: The word "newer" implies, that the files are closer to the current date. The oldest commit is the first ever commit, right? I would therefore expect the following output, when I filter for commit newer than `e91e087`:
```
➜ kiri ls -n e91e087
     1	a7bb19c    2024-04-02 15:07:25    Leandro Heck    Update to Kicad 8 (HEAD -> main, origin/main, origin/HEAD)
     2	7aebd2e    2023-03-22 16:09:20    Leandro Heck    Update to Kicad 7
     3	5be11dd    2021-12-29 14:58:15    Leandro Heck    Update 3d models dir variable (selected_layers)
     4	d0628e0    2021-12-29 14:54:10    Leandro Heck    Convert project to Kicad v6
     5	010b611    2021-12-29 14:25:14    Leandro Heck    Update using kicad 5.1.12 (tag: 1.0)
     6	20e163c    2021-02-09 10:53:38    Leandro Heck    Remove 3Ds, fix 3V3 and 1V8 switched labels
     7	f35d811    2021-01-23 11:23:21    Leandro Heck    Align filled zones on 0.5 mm grid
     8	0a59275    2021-01-23 11:14:10    Leandro Heck    Update copper areas
     9	e91e087    2021-01-22 18:28:56    Leandro Heck    Final touches with the labels
```

The commits `89f22a1`, `b59502b` and `224bb02` are older than `e91e087`, but in the current implementation those are the result for *newer*. May be there is another way of understanding "older" and "newer" but I can't think of any other meaning.

leoheck, 2025-01-30:
Hm, maybe it is better to understand if I use the superlative newest/coldest (commit of the list of commits that are going to be compared) instead of newer/older.

semiversus, 2025-01-30:
Yes, newest and oldest would make sense in the way it's implemented.

leoheck, 2025-01-30:
Alright, thank you. The fix was pushed. 
I am closing this issue, but feel free to reopen it or open another one if you still have problems to discuss.