leoheck/kiri #129: KiCad-Diff and dots in filenames

hp-pepster, 2025-03-03:
Hi,

thank you for your very useful project.

Just discovered it but the PCB diff did not work. After some digging, the source of the problem points to KiCad-Diff.

PCB-Files with dots in the name (e.g. proj_v2.2.kicad_pcb) do not play nicely with your difftool. I've already found a possible solution. Do you want me to open a PR or shall I post my changes here? It's just 3 lines of code.

leoheck, 2025-03-03:
Hey. Yeah that might happen eventually. If you have the solution, sure, open up the PR, please.

leoheck, 2025-03-04:
Hi @hp-pepster 
I could manage to find the issue you pointed out and fix the problem.
Can you test it, please? You will have to update the kicad-diff and also kiri, I believe.
I tested here, rewriting the history of a board I have.

Using this command
```bash
git filter-branch --tree-filter 'rename "s/board\./board_1.2v./g" * || true' -- --all
```

Anyway, you can use the same command if just want to check how kiri will work, by removing the point like this
Do this on a copy of the repo to avoid losing your data. But the command is safe.
```bash
git filter-branch --tree-filter 'rename "s/board_1\.2/board_./g" * || true' -- --all
```

I am going to close the issue, but let me know if everything went well of if something bad hapenned.

closed 2025-03-04 by hp-pepster