leoheck/kiri #130: MacOS install_dependencies.sh failed

alextrical, 2025-03-05:
Just tried setting up KiRi on a MacBook M1, using the following command
`bash -c "$(curl -fsSL https://raw.githubusercontent.com/leoheck/kiri/main/install_dependencies.sh)"`

the step it failed for me on was installing `opam install -y git-unix` https://github.com/leoheck/kiri/blob/5ea0d59852c384d771fb53bbb3a405af7f7e0f95/install_dependencies.sh#L265
and resulted in the following error.
```
...
# To update the current shell environment, run: eval $(opam env)

<><> Updating package repositories ><><><><><><><><><><><><><><><><><><><><>  🐫 
[default] no changes from https://opam.ocaml.org
[NOTE] Package digestif is already installed (current version is 1.2.0).
[NOTE] Package lwt is already installed (current version is 5.9.0).
[NOTE] Package lwt_ppx is already installed (current version is 5.8.0).
[NOTE] Package cmdliner is already installed (current version is 1.3.0).
[NOTE] Package base64 is already installed (current version is 3.5.1).
[NOTE] Package sha is already installed (current version is 1.15.4).
[NOTE] Package tyxml is already installed (current version is 4.6.0).
[ERROR] Sorry, resolution of the request timed out.
        Try to specify a more precise request, use a different solver, or increase the allowed time by setting OPAMSOLVERTIMEOUT to a
        bigger value (currently, it is set to 60.0 seconds).
alextrical@Alextricals-MacBook-Pro ~ %
```

the fix for me was running `opam install -y git` prior to running `opam install -y git-unix`
This could be added to the script to help future Mac users from hitting this error.

Plese let me know if you think its worth me submitting a PR for this fix?

Kind regards
Ben J

alextrical, 2025-03-05:
It also looks like there is a missing dependancy of dos2unix, that can be solved by running
`brew install dos2unix`

leoheck, 2025-03-05:
Sure, it is a good addition to make it work for all. Please submit the PR.