leoheck/kiri #132: pcb plots not available with KiCad 9.0.1

frankalicious, 2025-04-02:
plotInvisibleText was removed with https://gitlab.com/kicad/code/kicad/-/commit/7cc2b47d2ebb9d08546ec228ef8d1f6c79dbc170

If you use kiri the pcb plots are empty but no error on the cli.  

If you run kidiff manually you see the error
```
$ ~/.local/share/kiri/submodules/KiCad-Diff/kidiff/kidiff.py test.kicad_pro
...
Exporting PCB layers with plot_kicad_pcb...
Traceback (most recent call last):
  File "/home/alex/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 274, in <module>
    processBoard(board_path, plot_dir, args.quiet, args.verbose, args.frame, args.numbers)
  File "/home/alex/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 109, in processBoard
    popt.SetPlotInvisibleText(False)
    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotInvisibleText'

Traceback (most recent call last):
  File "/home/alex/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 274, in <module>
    processBoard(board_path, plot_dir, args.quiet, args.verbose, args.frame, args.numbers)
  File "/home/alex/.local/share/kiri/submodules/KiCad-Diff/bin/../kidiff/plot_kicad_pcb.py", line 109, in processBoard
    popt.SetPlotInvisibleText(False)
    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetPlotInvisibleText'
```
As a workaround I removed `popt.SetPlotInvisibleText(False)` in plot_kicad_pcb.py
```
~/.local/share/kiri/submodules/KiCad-Diff$ git diff
diff --git a/kidiff/plot_kicad_pcb.py b/kidiff/plot_kicad_pcb.py
index e6338ab..b37e46d 100755
--- a/kidiff/plot_kicad_pcb.py
+++ b/kidiff/plot_kicad_pcb.py
@@ -106,7 +106,7 @@ def processBoard(board_path, plot_dir, quiet=1, verbose=0, plot_frame=0, id_only
     popt.SetSubtractMaskFromSilk(False)
     popt.SetPlotReference(True)
     popt.SetPlotValue(True)
-    popt.SetPlotInvisibleText(False)
+    # popt.SetPlotInvisibleText(False)
     popt.SetPlotFrameRef(plot_frame)
 
     # Kicad >= 8.0.0
```
Now it seems to work again. I can see the diff in kiri.

leoheck, 2025-04-02:
Thanks, @frankalicious. I've pushed a fix. Can you test it?

closed 2025-04-02 by leoheck

frankalicious, 2025-04-02:
Thanks. That was fast. With your fix it works like a charm.