leoheck/kiri #133: Printing, plotting, saving, pdf export?

topherbuckley, 2025-04-08:
Is there currently any way to export a given schematic diff to some format (e.g. pdf, svg, etc.)? I wanted to share a diff with a coworker that won't have kiri installed on their machine.

leoheck, 2025-04-08:
If the web browser is running, it is possible to access it with the IP of your machine instead of the localhost.

```http://IP:8080/web/index.html```

You can also share the .kiri folder (or anything inside the folder)
Everything is inside there.

topherbuckley, 2025-04-10:
I find the individual svg files, but it appears the overlay and colorization is done vis javascript or did I misunderstand something? 

I was hoping for a way to print/save these overlaid and colorized diffs. i.e. show my coworker the same thing I'm looking at via kiri, but without forcing them to install kiri to see them.

leoheck, 2025-04-10:
they don't need to install anything. you share the generated .kiri folder and the script to launch the webserver will be inside that folder.

closed 2025-04-16 by leoheck