leoheck/kiri #134: Text search ability within schematics/layouts

topherbuckley, 2025-04-10:
Guessing this isn't possible right now, but just wanted to check. If I have a net name that I want to see where it connects, in KiCAD I'd just do a text search with ctrl+f. This would be very helpful in the work I'm doing right now trying to debug why something doesn't work within a new commit but did in the past. Net names changed, and plenty of relocations, so tracking everything down visually is tough. 

Do the svg files contain actual text, or is everything just a bundle of vectors?

leoheck, 2025-04-10:
> Do the svg files contain actual text, or is everything just a bundle of vectors?

The svgs are stored inside the geberates .kiri folder.

topherbuckley, 2025-04-29:
@leoheck sorry for my slow reaction, but just checking that you read my question. I wasn't looking for the svg files. 😅